antmicro/jetson-nano-baseboard #1: where can find and download this linux system ?

cudnn, 2019-03-20:
where can find and download this linux system ?

pkatarzynski, 2020-04-29:
Most of the IO interfaces available on board are supported by the default BSP provided by Nvidia
See https://developer.nvidia.com/embedded/downloads for more details.

closed 2020-04-29 by pkatarzynski

MI-LA01, 2020-11-06:
@pkatarzynski  : Just brought one unit via capable robot for testing. When you said "Most of the IO.." can you elaborate a bit more on that ? What are the IOs not supported by the BSP if any ?

pkatarzynski, 2020-11-06:
You  will most likely need to adjust the BSP for camera (CSI) interfaces. The Ethernet, USB-C, HDMI, Displayport, M.2 (NVMe) are handled by the default JetPack provided by NVIDIA for the Jetson Nano or the Jetson Xavier NX

MI-LA01, 2020-11-06:
Any Guidance/doc to tune the BSP for the 2x 50-pin FFC and 10-pin FFC PCIe x1 ?