antmicro/jetson-nano-baseboard #3: PCB files

kfriesth, 2019-04-03:
Would you export the pcb schematic and pcb layout files to kicad or gerber file formats?

qume, 2020-03-17:
Perhaps a bit late, but there is a free online Altium viewer here:

[https://www.altium.com/viewer/](https://www.altium.com/viewer/)

Drag and drop the schdoc files, and it will show you the schematic, the BOM, and in the case of the PCB file, the layout

I'll be starting a KiCAD board for the nano soon, did you end up doing one? Any interest in sharing the project?

pkatarzynski, 2020-04-29:
We have a KiCad version of this board converted from Altium files. We hope to share this version soon.

qume, 2020-04-29:
@pkatarzynski great news, thanks very much. Looking forward to it

AdrianCGonzalez, 2020-06-11:
Would it be possible to export the Altium PCB files to ASCII format?

wgthompson, 2021-03-26:
Longshot but any chance we can get the files in eagle?

wild929, 2021-09-15:
I am using Altium Circuit Studio and am unable to import the PCB layout. Is there a way that you can save the file as  a PCB 5.0 Binary file?

Cheers...

pgielda, 2021-09-15:
Not sure if that helps but there is a branch with Kicad version: https://github.com/antmicro/jetson-nano-baseboard/tree/kicad-version

wild929, 2021-09-15:
> Not sure if that helps but there is a branch with Kicad version: https://github.com/antmicro/jetson-nano-baseboard/tree/kicad-version

Thanks. I am downloading KiCad now in the hopes I can convert the file to PCB Binary 5. I really don't want to have to get up to speed on a new tool.

wild929, 2021-11-08:
Perhaps a bit off topic but can you tell me if the TX2 NX 's USB3 (which is 3.0) and USB1 (2.0) port can be connected directly to a USB3.0 Type A port? Or does this connection require a controller / mux between SOM and port?

Thanks.

pkatarzynski, 2021-11-08:
It is possible to connect the USB lanes you mentioned directly to an USB 3.0 type-A port.

mgielda, 2022-05-07:
As the main branch has switched to KiCad now, this issue can now be closed.

closed 2022-05-07 by mgielda