antmicro/jetson-nano-baseboard #5: The new PCB doesn't match with the schematic

sunzhongmeng, 2020-03-05:
hello ,i found the new pcb add a M.2 sloat,but the i can't find it in schematics.

pkatarzynski, 2020-03-06:
Hi sunzhongmeng 

The M.2 connector (J10) has been defined in interfaces.SchDoc file which has been accidentally excluded from the main project file (PrjPcb). We have fixed the project file.
Thank you for spotting this.

closed 2020-03-06 by pkatarzynski

sunzhongmeng, 2020-03-06:
> Hi sunzhongmeng
> 
> The M.2 connector (J10) has been defined in interfaces.SchDoc file which has been accidentally excluded from the main project file (PrjPcb). We have fixed the project file.
> Thank you for spotting this.

Hello Piotr 
I fetch the project you have update,yes I see the "Interfaces.SchDoc",I'm sorry to talk about the SD/MMC card,the README file shows the board has a SD/MMC card(the assembly pdf also shows the J8),I didn't see it in the Project files(neither schematics or pcb layout file)

pkatarzynski, 2020-03-06:
This should now be resolved.