antmicro/jetson-nano-baseboard #7: HDMI pull down resistors

sunzhongmeng, 2020-03-09:
Hello ，the NVIDIA Jetson Nano Product Designe Guide  says "499Ω, 1% pull-downs must be disabled when Jetson Nano is off or in sleep mode to meet the HDMI VOFF requirement. The enable to the FET, enables the pull-downs when the HDMI interface is to be used"but this design delete all pull down resistors.does it work well?

pkatarzynski, 2020-03-09:
We have simplified some of the interfaces to gain more space for routing and make the PCB smaller. The board is meant to work as technology demonstrator and evaluation platform which stays in always-on mode. Hence satisfying HDMI VOFF was not our priority. The PCBs in rev. 1.3 were displayed at our booth during Embedded World 2020 in [camera related application](https://www.linkedin.com/posts/antmicro-ltd_jetsonnano-xaviernx-opensource-activity-6638028555051638785-Hmo-/)

sunzhongmeng, 2020-03-09:
> We have simplified some of the interfaces to gain more space for routing and make the PCB smaller. The board is meant to work as technology demonstrator and evaluation platform which stays in [...]

I see ，thank u.

closed 2020-03-09 by sunzhongmeng