antmicro/jetson-nano-baseboard #10: Power sequensing

gudozhni, 2020-03-31:
Hello!

This might not be an issue, but something in your design contradicts Nvidia information.
It would be great if you could clarify it.
1) The datasheet expressly states that no signals should be applied to any module IO pin until it's completely powered up. To obtain such behavior, the evalkit design enables carrier power supplies (3.3, 1.8) with SYS_RESET, which goes high when module finished power up process. In your design, these power supplies are enabled by POWER_GOOD of the main 5V DC/DC, so they are enabled at the same time as 5V_SYS. You did it like that because it JUST WORKS and you don't see any adverse effect or I'm missing something? 
2) According to the graph on the page 9 of the Jetson Nano Product Design Guide, in case of sudden power loss the main power supply should decay from 5V to 3V no faster than at least 10ms, to allow shutting down the module gracefully after servicing SHUTDOWN_REQ and forcing POWER_EN low. The three 47uF caps on the 5V rail in your design would only last for ~50us before the voltage drops to 3V. So the evident question here is - did you have any troubles with hard power offs? Did you test such cases?

I'd be very grateful for any info on the 2 points above!

Best,
Olex

pkatarzynski, 2020-03-31:
Hello Olex

Thank you for your contribution.

As for the 1. this has been already discussed. We are going to release a new PCB revision which uses 
SYS_RESET for starting the on-board peripherals. Nevertheless using POWER_GOOD to start the DC/DC worked for us so far as well. 

As for 2. we have already done a couple of technology demonstrators using this board. Theese included (multi-) camera setups showcased during the last Embedded World. Most of those setups were powered off by disconnecting the main power supply. This by no means affected the module or software running on it. We will take a look at the voltage decay - we will record this curve on oscilloscope to have a real-life example for further analysis.

gudozhni, 2020-04-01:
Hello Piotr,
thanks a lot for your explanations! that is sufficient for me.

closed 2020-04-01 by gudozhni