antmicro/jetson-nano-baseboard #12: Manufacturing of the board / readily availbale boards for sale?

careyer, 2020-05-20:
Hi there,

thank you very much for taking all the effort and time to make this incredible and compact carrier board. What would be the best way to get one produced (any recommendations of manufacturers that can easily process the available design files?) or maybe is there somebody arround with a ready made spare board that he wants to sell?

Thank you very much and sorry for poluting the issues section =)

sunrise-ceo, 2020-05-22:
Supporting the question above. Would love to buy a couple of ready-made baseboards as well, for testing at first, but, potentially, for mass production of our products. Are there any online retailers (or maybe factories, accepting orders for production) who have this baseboard in stock? Thanks in advance.

kalvdans, 2020-11-09:
Is this fixed by https://github.com/antmicro/jetson-nano-baseboard/commit/56bf3e57818304ef53ff78908ac0a93a50a09229, @pkatarzynski ?

osterwood, 2020-11-17:
Yep!  Baseboards are available for purchase on the Capable Robot website.

https://capablerobot.com/products/nx-baseboard/

Stock on the initial batch is nearly sold out.  A much larger batch is being made now and will take about a month produce.

mgielda, 2020-12-01:
Thanks for answering this, Chris @osterwood :)

Indeed Capable Robot is selling them now, so I consider the issue closed.

closed 2020-12-01 by mgielda