antmicro/jetson-nano-baseboard #15: Any update on Kicad files?

BlueWaterCrystal, 2020-09-22:
what stage are the Kicad files currently at?

#3 
> We have a KiCad version of this board converted from Altium files. We hope to share this version soon.

tkircher, 2020-11-17:
I'd like to know too. I'm interested in a board with this form factor, but stripped down to the SoM connector, DC power, M.2 connector, USB 3.0, and debug port. Would be nice if I could route that in four layers and get it made at OSHpark. ;)

vanbwodonk, 2020-11-19:
I'm waiting for this too.

ncx94, 2021-03-08:
Hi,

Definitely would be interrested in that too!

abhibhatia98, 2021-05-07:
Hello Everyone, I am working on the same project, and completed the schematic and PCB is going to be completed in 2 more days. Any one here who is experienced PCB designer and is interested in this project and who can review my PCB are welcome. I am ready to share project. It will be helpful for other also.

PS: I am not an experienced PCB designer for high speed designs.
Thanks

tkircher, 2021-05-07:
One big advantage of the Jetson SoM is that most of the critical high speed design is done for you.

ncx94, 2021-05-07:
> Hello Everyone, I am working on the same project, and completed the schematic and PCB is going to be completed in 2 more days. Any one here who is experienced PCB designer and is interested in this [...]

Awesome thank you! Not a professional either but I'd be happy to have a look at it! :)

pkatarzynski, 2021-05-07:
We are working on the KiCad version. The design files will be released soon.

ncx94, 2021-07-21:
> We are working on the KiCad version. The design files will be released soon.

Hi,

Any updates on that?

Thanks

pkatarzynski, 2021-08-18:
Please take a look at the branch https://github.com/antmicro/jetson-nano-baseboard/tree/kicad-version

closed 2021-08-18 by pkatarzynski