antmicro/jetson-nano-baseboard #16: Problem with the HDMI

baroselfdriving, 2021-01-21:
HI. we are working with a modified version of your board adding other peripherals. 
We have a problem with the HDMI. The board recognise correctly the MONITOR. But I don't have video, any monitor say: NO SIGNAL. But when I connect the HDMI cable the VNC adjust the resolution automatically with the new monitor.

I review all the design and is the HDMI area is exactly your design. 

I installed the last l4t version 4.5 and didn't work. I used the previous and when boot appear not good defined the NVIDIA LOGO. Now I downgrade other version and I can see distorted the image with the initial SETUP of UBUNTU.

some
[image]

ahmetcanack, 2021-04-26:
Are you using the correct SOM, I believe there are 2 versions of modules, [You can see the differences here.](https://www.arducam.com/nvidia-jetson-nano-b01-update-dual-camera/)

I have also seen people designing custom carrier boards with the help of the documentation shared by Nvidia.  They were having problems with the ethernet. There is a lot of confusion about this.  Maybe this has to do something with your problem.

Jslater85, 2021-08-18:
Hi, I have encountered similar problems as mentioned  by Gabriel.

Did you ever have any resolution to this issue with the HDMI?

Which SOM is the correct to use?

All the best!

Jake

baroselfdriving, 2021-08-18:
We finally redesign the hdmi lanes to prevent the issue, following the rules of the hdmi design and we used the jetson carrier board to solve many small issues.
About the firmware for 4 cameras, finally we design it based in the e3333 dts in the jetpack and we have it published in our website.

Best regards

Gabriel Giani

pkatarzynski, 2021-08-18:
Hi all, HDMI works properly on the original design and many derivatives we have been building. If you redesign the board yourself and you cannot get something to work on the derivative design, this is not something we can help with. If you need help to build your own board, feel free to reach out to us to use our commercial services.

closed 2021-08-18 by mgielda