antmicro/jetson-nano-baseboard #17: Routing layers copper pours

iDevMartin, 2021-01-29:
I noticed the signal/routing layers in the layout do not contain large GND copper pours, is there a reason it was not added to the design? Shouldn't it help with impedance control and shielding?

ajawamnet, 2021-04-01:
In some cases it can lead to uncontrolled CPW - coplanar wave guides -
https://en.wikipedia.org/wiki/Coplanar_waveguide

https://www.microwavejournal.com/blogs/1-rog-blog/post/24374-comparing-microstrip-and-grounded-coplanar-waveguide
In the newer versions of altium the simbeor LSM will "sort of" calculate these.   Use of CPW also requires careful consideration of via stiching to the contiguous  reference plane 

Regardless, using copper pours is typically not recommended except in certain situations.  As to using it for thieving note what Lee Ritchey states here:
https://ee-training.dk/pcb-manufacturing/copper-thieving-confusion.htm

" If the next layer down is a plane layer,
that is all that is needed. If the next layer down is a signal layer (buried
microstrip) thieving should not be placed over traces in the buried
layer."

I have to agree - I've designed over 3,000 PCB's of various complexities and the only times I've had to use copper pours was for either current requirements or CPW to reduce trace widths to match tiny RF connectors. And the latter is a pain due tot he stitching requirements

Broadcom has a nice, free calculator that has CPW:
https://www.broadcom.com/appcad
This is a later version of the original HP calculator. If you go into the CPW part you can see the effect of having copper pours edge coupling to traces. 

And using copper pours on internal layers can lead to split plane issues that cause discontinuities in reference to things like controlled impedance traces - ESP if you are using asymmetric striplines.

mgielda, 2022-05-07:
Since this was explained quite a while ago, I think we can close the issue.

closed 2022-05-07 by mgielda