antmicro/jetson-nano-baseboard #19: Use of below circuitry

abhibhatia98, 2021-04-26:
Hi I am just looking at your circuit, could you please help me to understand the use of below circuit in supply sheet

[image]

Thanks
Abhishek

ahmetcanack, 2021-04-26:
Hi, if you look at chapter 5.1 Power Supply and Sequencing in the jetson nano product design guide, you will see a power sequencing graph. Basically, power must not be delivered without control, for some power rails to be enabled, other rails or ok signals must be checked and validated. This circuit makes sure that these enable signals are generated correctly following the power sequencing requirements in the jetson nano product design guideline. I suggest you read the first paragraph in chapter 5.1. There is a nice explanation for all this. And also you might check out how nand gates work. In terms of schematic layout i find this particular layout a bit clustered. But its not impossible to understand. This 4x NAND IC could have been seperated as four individual schematic symbols each showing a NAND symbol. That way it could have been easier for all to read. Lastly this is a great project for individuals to create their custom , and you will find it easier if you go through the product design guide.

abhibhatia98, 2021-04-26:
Thanks for your response, it helps me a lot

closed 2021-04-26 by abhibhatia98