antmicro/jetson-nano-baseboard #21: Undetected CSI camera

atoaster, 2021-07-16:
Hi all,

I've been trying to use this board with my IMX219 camera (it works with the Jetson Nano DevKit) to no avail.

My setup is:

Jetson Baseboard [from capablerobots](https://capablerobot.com/products/nx-baseboard/) 
15pin CSI connector [from capablerobots](https://capablerobot.com/products/nx-baseboard/#tab-flex)
IMX219-AF [from arducam](https://www.arducam.com/product/b0181-arducam-imx219-af-programmable-auto-focus-camera-module-nvidia-jetson-nano/)

Am running latest L4T and have installed JetPack via the command line (sudo apt install nvidia-jetpack)

I've got the board setup with the camera connected to Camera FFC #1, but it never gets detected by `nvgstcapture-1.0`. Here's the output from `nvgstcapture-1.0`:

`<main:4670> iterating capture loop ....
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:645 No cameras available

(nvgstcapture-1.0:6251): GStreamer-CRITICAL **: 13:21:08.592: gst_mini_object_set_qdata: assertion 'object != NULL' failed`

So clearly it just can't see the connected camera.

Doesn't the 15pin cable effectively simulate the same 15pin connection that the Nano DevKit has? What else could be different between these two boards?

kgugala, 2021-07-16:
Hi @atoaster,

One of the differences between the devkit and our board is that on our board the cameras I2C lines are connected through an I2C mux:

[image]

You need to modify your devicetree to express this. See the patches we provide for Allied Vision cameras. Here https://github.com/antmicro/jetson-nano-baseboard/blob/master/linux-patches/32.4.4.patch#L131 we define the extender and here https://github.com/antmicro/jetson-nano-baseboard/blob/master/linux-patches/32.4.4.patch#L144 we attach the camera driver (as a device one of the child buses of the I2C mux). You need to do similar modification to you DTS and the camera should get detected.

atoaster, 2021-07-21:
Hi @kgugala, thanks for getting back to me!

I don't really know what I'm doing so bear with me...

I'm trying to edit the existing `sources/hardware/nvidia/platform/t19x/jakku/kernel-dts/common/tegra194-camera-rbpcv2-imx219.dtsi` file, since it has the word camera in it.

Since I'm using the 15pin flex cable from CapableRobots, I shouldn't have to change any pin configurations, right? In other words, I should be able to use the regular device tree .dts configuration, but put them behind an `i2c` node (the same one from your av patches)?

Here's the existing dtsi (`tegra194-camera-rbpcv2-imx219.dtsi`):

```
/ {
	host1x {
             ...
	};

	cam_i2cmux {
		i2c_0:i2c@0 {
			imx219_cam0: rbpcv2_imx219_a@10 {
                                ...
			};
		};
		i2c_1: i2c@1 {
			imx219_cam1: rbpcv2_imx219_c@10 {
                                ...
			};
		};
	};
```

I'm going to try this :

```
/ {
	host1x {
                ...
	};
	i2c@3180000 {
                avt_pca9548: avt_pca9548@70 {
			compatible = "nxp,pca9548";
			status = "okay";
			reg = <0x70>;
			#address-cells = <1>;
			#size-cells = <0>;
			vcc-supply = <&battery_reg>;
			force_bus_start = <CAMERA_I2C_MUX_BUS(0)>;
			i2c@2 {
				reg = <2>;
				i2c-mux,deselect-on-exit;
				#address-cells = <1>;
				#size-cells = <0>;
                                imx219_cam0: rbpcv2_imx219_a@10 {
                                        ...
                                }
                       }
               }
       }
}
```
I'm assuming I won't have to do anything to the properties of the `i2c@2` node (hopefully)

Will report back, hopefully this can help anyone else using this board :)

atoaster, 2021-07-26:
Hi @kgugala,

Is the patch meant for the Tegra Xavier SoC? I can't seem to find the `0x3180000` address in the Tegra X1 Technical Reference Manual.

kgugala, 2021-07-29:
Tegra X1 is not meant to work with this baseboard. The patches are for Jetson Xavier NX (see https://github.com/antmicro/jetson-nano-baseboard#bsp)

atoaster, 2021-07-30:
Woops, I was referencing the wrong document! I'm actually using a Jetson Nano.

I've actually gotten the i2cmux working with the following device tree nodes (Apparently the i2c node in the Nano is /host1x/i2c@546c0000)

So now my tree looks like this:
```
    host1x {
		i2c@546c0000 {
			#address-cells = <0x01>;
			#size-cells = <0x00>;
			compatible = "nvidia,tegra210-vii2c";
			reg = <0x00 0x546c0000 0x00 0x34000>;
			iommus = <0x30 0x12>;
			interrupts = <0x00 0x11 0x04>;
			scl-gpio = <0x5b 0x92 0x00>;
			sda-gpio = <0x5b 0x93 0x00>;
			status = "okay";
			clocks = <0x26 0xd0 0x26 0x51 0x26 0x1c>;
			clock-names = "vii2c\0i2cslow\0host1x";
			resets = <0x26 0xd0>;
			reset-names = "vii2c";
			clock-frequency = <0x61a80>;
			bus-pullup-supply = <0x47>;
			avdd_dsi_csi-supply = <0x3b>;
			linux,phandle = <0xad>;
			phandle = <0xad>;

			tca9548@70 {
				compatible = "nxp,pca9548";
				status = "okay";
				reg = <0x70>;
				#address-cells = <1>;
				#size-cells = <0>;
				vcc-supply = <0x4c>;
				force_bus_start = <0x1E>;
				i2c@0 {
					reg = <0>;
					i2c-mux,deselect-on-exit;
					rbpcv2_imx219_a@10 {
						compatible = "nvidia,imx219";
						reg = <0x10>;
						devnode = "video0";
						physical_w = "3.680";
						physical_h = "2.760";
						sensor_model = "imx219";
						use_sensor_mode_id = "true";
						status = "okay";
						reset-gpios = <0x5b 0x97 0x00>;
						linux,phandle = <0xbe>;
						phandle = <0xbe>;
                                                mode0 { ...
                                                };
                                               ...
                                       };
    };
```

Which allows for the i2c mux to be detected by i2cdetect:
```
i2cdetect -l
i2c-35  i2c             i2c-6-mux (chan_id 5)                   I2C adapter
i2c-3   i2c             7000c700.i2c                            I2C adapter
i2c-33  i2c             i2c-6-mux (chan_id 3)                   I2C adapter
i2c-1   i2c             7000c400.i2c                            I2C adapter
i2c-31  i2c             i2c-6-mux (chan_id 1)                   I2C adapter
i2c-6   i2c             Tegra I2C adapter                       I2C adapter
i2c-36  i2c             i2c-6-mux (chan_id 6)                   I2C adapter
i2c-4   i2c             7000d000.i2c                            I2C adapter
i2c-34  i2c             i2c-6-mux (chan_id 4)                   I2C adapter
i2c-2   i2c             7000c500.i2c                            I2C adapter
i2c-32  i2c             i2c-6-mux (chan_id 2)                   I2C adapter
i2c-0   i2c             7000c000.i2c                            I2C adapter
i2c-30  i2c             i2c-6-mux (chan_id 0)                   I2C adapter
i2c-37  i2c             i2c-6-mux (chan_id 7)                   I2C adapter
i2c-5   i2c             7000d100.i2c                            I2C adapter

```

But during boot I'm still getting

` imx219 7-0010: imx219_board_setup: error during i2c read probe (-121)`

Do you know `i2c@` number I should be using for the `tca9548`? I've assumed the 15pin flex cable from capablerobots uses the first one, ie `i2c@0`, but it clearly isn't working for me. 

I noticed the patches use `i2c@2` and `i2c@3`, is that significant only to the Allied Vision cameras?

Thanks again for all your help

osterwood, 2021-07-30:
The FFC on the narrow side of the PCB is J7, which has i2c@2 and i2c@3.  The FFC on the long side of the PCB is J6, which has i2c@0 and i2c@1.  If you've connected the 15 pin FFC cable to J7, your imx219 node should be under tca9548@70 -> i2c@2 

[image]

atoaster, 2021-08-02:
@osterwood Thanks for helping me out!

I've now got the camera seemingly being detected, but I'm having a hard time getting it working. It's hitting me with a 
```
[    1.843288] imx219 32-0010: i2c bus regbase unavailable
[    1.848560] imx219 32-0010: Could not initialize sensor properties.
[    1.854869] imx219 32-0010: Failed to initialize imx219
[    1.860117] imx219 32-0010: tegra camera driver registration failed

```

I know it's a long shot, but do you have a working `tca9548@70/i2c@x/imx219` dts node that I can copy? I've tried copying the default `imx219` nodes into the `i2c@2` tree and they all seem to throw this exact error. 

I feel like the solution is so close!

atoaster, 2021-08-04:
Update: I've finally gotten the DTB working! There was an issue where running a `sudo apt update && sudo apt upgrade` on L4T would replace the DTB, so I had to get it working like follows:

1. Flash Nano
2. copy signed *.dtb from /boot (whichever is relevant to your board version)
3. `sudo apt update && sudo apt upgrade`
4. copy signed dtb back to /boot
5. add FDT /boot/your-custom-device-tree-binary.dtb to the /boot/extlinux/extlinux.conf

But now for some reason `nvgstcapture` will start but it doesn't display the camera feed, not entirely sure why that is, probably has something to do with the DTS I guess

atoaster, 2021-08-04:
Ok, final comment! Looks like `nvgstcapture` doesn't play nice with fps less than 30 sometimes. Here's what I use to display the stream from the camera.

`gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM), width=3280, height=2464, format=NV12, framerate=(fraction)21/1' ! nvvidconv ! xvimagesink`

If anyone wants the DTS I used for making an IMX219 work with this baseboard and a Jetson Nano, here ya go: 
https://gist.github.com/atoaster/15a533de112f50cc0ee8299dcef8c859

closed 2021-08-04 by atoaster