antmicro/jetson-nano-baseboard #22: Jetson Xavier NX board boot problem

citrusboy1997, 2021-08-12:
According to your carrier board design, I built a custom version and prototyped the board. Jetson nano module booted successfully on my own board, but Jetson Xavier NX  module didn't. Is Xavier NX not able to boot directly on your design but requires more settings?

osterwood, 2021-08-13:
The published design works with Nano, Xavier NX, and TX2 NX SOMs from NVIDIA.  Without knowing the changes you made it is difficult to know why your design works with a Nano and not the NX.

mgielda, 2021-08-18:
Hi @citrusboy1997, if you redesign the board yourself and you cannot get something to work on the derivative design, this is not something we can help with. If you need help to build your own board or running software on it, feel free to reach out to us to use our commercial services.

closed 2021-08-18 by mgielda