antmicro/jetson-nano-baseboard #23: Attempting to flash a TX2 NX SoM throws UnicodeDecodeError

atoaster, 2021-08-31:
Hi All, it's me again

I've been toying with using a TX2 NX SoM on your amazing baseboard, but I am getting a UnicodeDecodeError when trying to flash it with the L4T tools.

Here is the full output from `nvautoflash` in JetPack 4.6 (note that this folder was made by the sdkmanager):

```
*** Checking ONLINE mode ... OK.
*** Checking target board connection ... 1 connections found.
*** Reading ECID ... FUSELEVEL=fuselevel_production hwchipid=0x18 bootauth=NS
*** Reading EEPROM ... "/home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/tegraflash.py" --chip 0x18 --applet "/home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/mb1_recovery_prod.bin" --skipuid --cmd "dump eeprom boardinfo cvm.bin; dump eeprom baseinfo bbd.bin; reboot recovery" 
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0029 ] Generating RCM messages
[   0.0051 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x18 0 --download rcm /home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/mb1_recovery_prod.bin 0 0
[   0.0062 ] RCM 0 is saved as rcm_0.rcm
[   0.0065 ] RCM 1 is saved as rcm_1.rcm
[   0.0066 ] List of rcm files are saved in rcm_list.xml
[   0.0066 ] 
[   0.0066 ] Signing RCM messages
[   0.0093 ] tegrasign_v3.py --key None --list rcm_list.xml --pubkeyhash pub_key.key
[   0.0094 ] Assuming zero filled SBK key
[   0.0167 ] Copying signature to RCM mesages
[   0.0195 ] tegrarcm_v2 --chip 0x18 0 --updatesig rcm_list_signed.xml
[   0.0208 ] 
[   0.0209 ] Boot Rom communication
[   0.0231 ] tegrarcm_v2 --chip 0x18 0 --rcm rcm_list_signed.xml --skipuid
[   0.0242 ] RCM version 0X180001
[   0.0253 ] Boot Rom communication completed
[   1.0318 ] 
[   2.0365 ] tegrarcm_v2 --isapplet
[   2.0391 ] Applet version 01.00.0000
[   2.0422 ] 
[   2.0463 ] Retrieving EEPROM data
[   2.0466 ] tegrarcm_v2 --oem platformdetails eeprom cvm /home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/cvm.bin
[   2.0489 ] Applet version 01.00.0000
[   2.0526 ] Saved platform info in /home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/cvm.bin
[   2.1251 ] 
[   2.1252 ] Generating RCM messages
[   2.1277 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x18 0 --download rcm /home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/mb1_recovery_prod.bin 0 0
[   2.1288 ] RCM 0 is saved as rcm_0.rcm
[   2.1293 ] RCM 1 is saved as rcm_1.rcm
[   2.1294 ] List of rcm files are saved in rcm_list.xml
[   2.1294 ] 
[   2.1294 ] Signing RCM messages
[   2.1326 ] tegrasign_v3.py --key None --list rcm_list.xml --pubkeyhash pub_key.key
[   2.1327 ] Assuming zero filled SBK key
[   2.1417 ] Copying signature to RCM mesages
[   2.1444 ] tegrarcm_v2 --chip 0x18 0 --updatesig rcm_list_signed.xml
[   2.1464 ] 
[   2.1465 ] Boot Rom communication
[   2.1491 ] tegrarcm_v2 --chip 0x18 0 --rcm rcm_list_signed.xml
[   2.1504 ] BootRom is not running
[   7.4048 ] 
[   8.4108 ] tegrarcm_v2 --isapplet
[   8.4139 ] Applet version 01.00.0000
[   8.4173 ] 
[   8.4218 ] Retrieving EEPROM data
[   8.4221 ] tegrarcm_v2 --oem platformdetails eeprom cvb /home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/bbd.bin
[   8.4250 ] Applet version 01.00.0000
[   8.4435 ] 00Traceback (most recent call last):
  File "/home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/tegraflash.py", line 1296, in <module>
    tegraflash_run_commands()
  File "/home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/tegraflash.py", line 1155, in tegraflash_run_commands
    interpreter.onecmd(command)
  File "/usr/lib/python3.9/cmd.py", line 217, in onecmd
    return func(arg)
  File "/home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/tegraflash.py", line 785, in do_dump
    tegraflash_dump(exports, args)
  File "/home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 1370, in tegraflash_dump
    tegraflash_dumpeeprom(args, dump_args[1:])
  File "/home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 1448, in tegraflash_dumpeeprom
    run_command(command)
  File "/home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 202, in run_command
    log = print_process(process, enable_print)
  File "/home/atoaster/nvidia/nvidia_sdk/JetPack_4.6_Linux_JETSON_TX2_TARGETS/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 178, in print_process
    for string in process.communicate()[0].decode('utf-8').split('\n'):
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa4 in position 16: invalid start byte
--- Error: Reading board information failed.

```

I've gone down the Nvidia Forum rabbit hole and it led me [here](https://forums.developer.nvidia.com/t/invalid-start-byte-when-flashing-tx2-nx-on-custom-board/185998/3?u=an_actual_toaster), indicating I need some kinda board configuration file "from the vendor". 

Does this file exist anywhere? Or do I need to make it? Has anyone succeeded in flashing JetPack 4.6 on a TX2 NX SoM with this baseboard?

osterwood, 2021-08-31:
I've not tested 4.6, but the CLI workflow does work for 4.5.1 (e.g L4T 32.5.1).

```
wget https://developer.nvidia.com/embedded/l4t/r32_release_v5.1/r32_release_v5.1/t186/tegra186_linux_r32.5.1_aarch64.tbz2
wget https://developer.nvidia.com/embedded/l4t/r32_release_v5.1/r32_release_v5.1/t186/tegra_linux_sample-root-filesystem_r32.5.1_aarch64.tbz2

export L4T_RELEASE_PACKAGE='tegra186_linux_r32.5.1_aarch64.tbz2'
export SAMPLE_FS_PACKAGE='tegra_linux_sample-root-filesystem_r32.5.1_aarch64.tbz2'

tar xvf ${L4T_RELEASE_PACKAGE}
cd Linux_for_Tegra/rootfs/
sudo tar xvpf ../../${SAMPLE_FS_PACKAGE}
cd ..

sudo ./apply_binaries.sh
sudo ./flash.sh jetson-xavier-nx-devkit-tx2-nx internal
```

I can try the above with the proper links for 4.6, but likely not in the next day or two.  If you want to attempt, the L4T archive has the URLs:

https://developer.nvidia.com/embedded/jetpack-archive
https://developer.nvidia.com/embedded/linux-tegra-r3261

atoaster, 2021-08-31:
Welp, just tried 4.5.1 using the `sdkmanager` and it worked fine, whereas 4.6 did not. 

I guess it's just something to do with 4.6 specifically?

Will leave this issue open if you find anything but otherwise my problem is solved by just using 4.5.1 :)

closed 2023-03-11 by atoaster