antmicro/jetson-nano-baseboard #24: USB C not working at boot

vdsmax, 2021-11-09:
Hi, I have an issue with the USB C from the board. When I boot my Xavier NX card with the board, I am not able to use a keyboard or a mouse. The card seems to boot normaly and I have the hdmi port which is working.
How can I available the USB C to work on boot ?
I am able to debug the board through an ssh connection.
The Xavier NX card works well on the development kit.

Here are the logs with the dmesg command. It seems that the USB C is working at the beginning of the boot (1.837254), then it is closed (11.972174):

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.140-tegra (buildbrain@mobile-u64-3357) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05) ) #1 SMP PREEMPT Thu Jun 25 21:22:12 PDT 2020
[    0.000000] Boot CPU: AArch64 Processor [4e0f0040]
[    0.000000] OF: fdt:memory scan node memory, reg size 48,
[    0.000000] OF: fdt: - 80000000 ,  2c000000
[    0.000000] OF: fdt: - ac200000 ,  44600000
[    0.000000] OF: fdt: - 100000000 ,  180000000
[    0.000000] earlycon: tegra_comb_uart0 at MMIO32 0x000000000c168000 (options '')
[    0.000000] bootconsole [tegra_comb_uart0] enabled
[    0.000000] Found tegra_fbmem: 00140000@a069c000
[    0.000000] Found lut_mem: 00002008@a0696000
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb1_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb1_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb2_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb2_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb3_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb3_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: reserved mem: initialized node ramoops_carveout, compatible id nvidia,ramoops
[    0.000000] OF: reserved mem: initialized node vpr-carveout, compatible id nvidia,vpr-carveout
[    0.000000] OF: reserved mem: initialized node grid-of-semaphores, compatible id nvidia,gosmem
[    0.000000] cma: Reserved 64 MiB at 0x00000000c2000000
[    0.000000] On node 0 totalpages: 2032317
[    0.000000]   DMA zone: 7179 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 459453 pages, LIFO batch:31
[    0.000000]   Normal zone: 24576 pages used for memmap
[    0.000000]   Normal zone: 1572864 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.1
[    0.000000] percpu: Embedded 25 pages/cpu @ffffffc1ffd05000 s61656 r8192 d32552 u102400
[    0.000000] pcpu-alloc: s61656 r8192 d32552 u102400 alloc=25*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Speculative Store Bypass Disable mitigation not required
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 2000562
[    0.000000] Kernel command line: console=ttyTCU0,115200 video=tegrafb no_console_suspend=1 earlycon=tegra_comb_uart,mmio32,0x0c168000 gpt tegra_fbmem=0x140000@0xa069c000 lut_mem=0x2008@0xa0696000 usbcore.old_scheme_first=1 tegraid=19.1.2.0.0 maxcpus=6 boot.slot_suffix=_b boot.ratchetvalues=0.4.2 vpr_resize sdhci_tegra.en_boot_part_access=1    quiet root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyTCU0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0
[    0.000000] log_buf_len individual max cpu contribution: 32768 bytes
[    0.000000] log_buf_len total cpu_extra contributions: 163840 bytes
[    0.000000] log_buf_len min size: 32768 bytes
[    0.000000] log_buf_len: 262144 bytes
[    0.000000] early log buf free: 29176(89%)
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Memory: 7189096K/8129268K available (15294K kernel code, 2934K rwdata, 6640K rodata, 8576K init, 609K bss, 186508K reserved, 753664K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff8008080000 - 0xffffff8008f70000   ( 15296 KB)
[    0.000000]     .rodata : 0xffffff8008f70000 - 0xffffff8009600000   (  6720 KB)
[    0.000000]       .init : 0xffffff8009600000 - 0xffffff8009e60000   (  8576 KB)
[    0.000000]       .data : 0xffffff8009e60000 - 0xffffff800a13d808   (  2935 KB)
[    0.000000]        .bss : 0xffffff800a13d808 - 0xffffff800a1d5f7c   (   610 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf00000000 - 0xffffffbf08000000   (   128 MB actual)
[    0.000000]     memory  : 0xffffffc000000000 - 0xffffffc200000000   (  8192 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 64.
[... 821 lines cut ...]
[   13.631109] zram3: detected capacity change from 0 to 678977536
[   13.655910] Adding 663060k swap on /dev/zram3.  Priority:5 extents:1 across:663060k SS
[   13.659468] zram4: detected capacity change from 0 to 678977536
[   13.678185] Adding 663060k swap on /dev/zram4.  Priority:5 extents:1 across:663060k SS
[   13.684328] zram5: detected capacity change from 0 to 678977536
[   13.700451] Adding 663060k swap on /dev/zram5.  Priority:5 extents:1 across:663060k SS
[   15.223096] Bridge firewalling registered
[   15.281285] nf_conntrack version 0.5.0 (65536 buckets, 262144 max)
[   15.467107] eqos 2490000.ether_qos eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   15.467738] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   15.554175] Netfilter messages via NETLINK v0.30.
[   15.558261] ctnetlink v0.93: registering with nfnetlink.
[   15.761301] IPv6: ADDRCONF(NETDEV_UP): docker0: link is not ready
[   16.301735] tegradc 15200000.nvdisplay: blank - powerdown
[   16.358312] extcon-disp-state external-connection:disp-state: cable 47 state 0 already set.
[   16.358320] Extcon AUX1(HDMI) disable
[   16.382211] tegra_nvdisp_handle_pd_disable: Powergated Head1 pd
[   16.382736] tegra_nvdisp_handle_pd_disable: Powergated Head0 pd
[   16.383016] tegradc 15200000.nvdisplay: unblank
[   16.384081] tegra_nvdisp_handle_pd_enable: Unpowergated Head0 pd
[   16.384196] tegra_nvdisp_handle_pd_enable: Unpowergated Head1 pd
[   16.386217] Parent Clock set for DC plld2
[   16.390694] tegradc 15200000.nvdisplay: hdmi: tmds rate:148500K prod-setting:prod_c_hdmi_111m_223m
[   16.413073] tegradc 15200000.nvdisplay: unblank
[   16.413089] tegradc 15210000.nvdisplay: blank - powerdown

pkatarzynski, 2021-11-09:
The USB-C should work in host mode by default while using a standard BSP offered by NVIDIA for this SoM.
Could you paste the output of a lsusb command after the system is booted? 
Could you paste the output of sudo dmesg -w after connecting a USB device to the port?

osterwood, 2021-11-10:
It would be a bit clearer if you could:
- boot the machine
- wait 10 seconds or so
- plug in USB devices
- then run `dmesg`

This way the USB messages will all be clustered at the bottom of the log.

It looks like your downstream hub is seen -- look for "USB hub found" in the debug into you've provided.  Not sure why it isn't showing up via `lsusb`

For reference, this is expected output with some USB 2 hub & devices attached:
```
nvidia@nxdev:~$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 480M
    |__ Port 2: Dev 5, If 0, Class=Hub, Driver=hub/3p, 480M
        |__ Port 2: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
```

The "device usb0 entered promiscuous mode -> link is not ready" is from the USB0 network device (which is used for USB network cards). That is expected output if there are no USB network devices.

vdsmax, 2021-11-10:
I did as you asked. I unplugged everything from my board except the HDMI cable. Then after 34s, I plugged the ethernet cable to be able to connect by ssh and have the logs. Just after plugging the ethernet cable, I plugged the usb device. It doesn't appear with the `dmesg` command:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.140-tegra (buildbrain@mobile-u64-3357) (gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05) ) #1 SMP PREEMPT Thu Jun 25 21:22:12 PDT 2020
[    0.000000] Boot CPU: AArch64 Processor [4e0f0040]
[    0.000000] OF: fdt:memory scan node memory, reg size 48,
[    0.000000] OF: fdt: - 80000000 ,  2c000000
[    0.000000] OF: fdt: - ac200000 ,  44600000
[    0.000000] OF: fdt: - 100000000 ,  180000000
[    0.000000] earlycon: tegra_comb_uart0 at MMIO32 0x000000000c168000 (options '')
[    0.000000] bootconsole [tegra_comb_uart0] enabled
[    0.000000] Found tegra_fbmem: 00140000@a069c000
[    0.000000] Found lut_mem: 00002008@a0696000
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb1_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb1_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb2_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb2_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb3_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: fdt:Reserved memory: failed to reserve memory for node 'fb3_carveout': base 0x0000000000000000, size 0 MiB
[    0.000000] OF: reserved mem: initialized node ramoops_carveout, compatible id nvidia,ramoops
[    0.000000] OF: reserved mem: initialized node vpr-carveout, compatible id nvidia,vpr-carveout
[    0.000000] OF: reserved mem: initialized node grid-of-semaphores, compatible id nvidia,gosmem
[    0.000000] cma: Reserved 64 MiB at 0x00000000c2000000
[    0.000000] On node 0 totalpages: 2032317
[    0.000000]   DMA zone: 7179 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 459453 pages, LIFO batch:31
[    0.000000]   Normal zone: 24576 pages used for memmap
[    0.000000]   Normal zone: 1572864 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.1
[    0.000000] percpu: Embedded 25 pages/cpu @ffffffc1ffd05000 s61656 r8192 d32552 u102400
[    0.000000] pcpu-alloc: s61656 r8192 d32552 u102400 alloc=25*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Speculative Store Bypass Disable mitigation not required
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 2000562
[    0.000000] Kernel command line: console=ttyTCU0,115200 video=tegrafb no_console_suspend=1 earlycon=tegra_comb_uart,mmio32,0x0c168000 gpt tegra_fbmem=0x140000@0xa069c000 lut_mem=0x2008@0xa0696000 usbcore.old_scheme_first=1 tegraid=19.1.2.0.0 maxcpus=6 boot.slot_suffix=_b boot.ratchetvalues=0.4.2 vpr_resize sdhci_tegra.en_boot_part_access=1    quiet root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyTCU0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0
[    0.000000] log_buf_len individual max cpu contribution: 32768 bytes
[    0.000000] log_buf_len total cpu_extra contributions: 163840 bytes
[    0.000000] log_buf_len min size: 32768 bytes
[    0.000000] log_buf_len: 262144 bytes
[    0.000000] early log buf free: 29176(89%)
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Memory: 7189096K/8129268K available (15294K kernel code, 2934K rwdata, 6640K rodata, 8576K init, 609K bss, 186508K reserved, 753664K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB)
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB)
[    0.000000]       .text : 0xffffff8008080000 - 0xffffff8008f70000   ( 15296 KB)
[    0.000000]     .rodata : 0xffffff8008f70000 - 0xffffff8009600000   (  6720 KB)
[    0.000000]       .init : 0xffffff8009600000 - 0xffffff8009e60000   (  8576 KB)
[    0.000000]       .data : 0xffffff8009e60000 - 0xffffff800a13d808   (  2935 KB)
[    0.000000]        .bss : 0xffffff800a13d808 - 0xffffff800a1d5f7c   (   610 KB)
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB)
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB)
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB maximum)
[    0.000000]               0xffffffbf00000000 - 0xffffffbf08000000   (   128 MB actual)
[    0.000000]     memory  : 0xffffffc000000000 - 0xffffffc200000000   (  8192 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[... cut ...]
[   12.385275] tegradc 15210000.nvdisplay: blank - powerdown
[   34.814597] eqos 2490000.ether_qos eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   34.815265] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

With the `lsusb -t` command, i have only this:
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 480M
```

I don't understand why it sees a usb hub at [1.768934]. It should not be the case. Everything were unplugged.

osterwood, 2021-11-10:
My mistake -- no idea why those "Hub found" messages are there, but a test setup here has the same output when booting when nothing is connected to USB.

```
[    1.809793] usbcore: registered new interface driver uas
[    1.809844] usbcore: registered new interface driver usb-storage
[    1.809919] usbcore: registered new interface driver usbserial
[    1.811659] tegra-xusb 3610000.xhci: Firmware timestamp: 2019-07-24 05:47:34 UTC, Version: 60.06 release
[    1.811737] tegra-xusb 3610000.xhci: xHCI Host Controller
[    1.811786] tegra-xusb 3610000.xhci: new USB bus registered, assigned bus number 1
[    1.812862] tegra-xusb 3610000.xhci: hcc params 0x0184ff25 hci version 0x110 quirks 0x00050810
[    1.812972] tegra-xusb 3610000.xhci: irq 463, io mem 0x03610000
[    1.813442] tegra_nvdisp_handle_pd_disable: Powergated Head1 pd
[    1.813471] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    1.813508] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.813516] usb usb1: Product: xHCI Host Controller
[    1.813521] usb usb1: Manufacturer: Linux 4.9.140-tegra xhci-hcd
[    1.813527] usb usb1: SerialNumber: 3610000.xhci
[    1.814646] hub 1-0:1.0: USB hub found
[    1.814740] hub 1-0:1.0: 4 ports detected
[    1.814851] tegra_nvdisp_handle_pd_disable: Powergated Head0 pd
[    1.815849] tegra-xusb 3610000.xhci: xHCI Host Controller
[    1.815878] tegra-xusb 3610000.xhci: new USB bus registered, assigned bus number 2
[    1.815897] tegra-xusb 3610000.xhci: Host supports USB 3.1 Enhanced SuperSpeed
[    1.816440] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003
[    1.816450] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.816459] usb usb2: Product: xHCI Host Controller
[    1.816465] usb usb2: Manufacturer: Linux 4.9.140-tegra xhci-hcd
[    1.816471] usb usb2: SerialNumber: 3610000.xhci
[    1.817911] hub 2-0:1.0: USB hub found
[    1.817993] hub 2-0:1.0: 4 ports detected
[    1.819015] tegra-xudc-new 3550000.xudc: PMQOS CPU boost enabled
[    1.819147] tegra-xudc-new 3550000.xudc: device count: 1
[    1.821387] tegra-xudc-new 3550000.xudc: vbus state: 0
[    1.821407] tegra-xudc-new 3550000.xudc: Initialize boost_cpufreq work
[    1.821477] tegra-xudc-new 3550000.xudc: entering ELPG
[    1.821912] tegra-xudc-new 3550000.xudc: entering ELPG done
```

What USB device(s) are you attaching?  Is there another USB-C host which you can use to verify that the cable is good?  

There may be something wrong with your board, but we do test the USB-C on each and every Baseboard prior to shipment, so I'm trying to understand if there might be some other variable I'm not accounting for.

vdsmax, 2021-11-10:
Thank you for your suggestion. I tried another USB hub using a USB 3.0 connection and it worked.
The former one I used was using a USB 2.0 connection.
I think the issue comes from this difference.  Is that normal a USB 2.0 connection isn't working  with the board ?

Here are the logs with the command dmesg after plugin a USB 3.0 device:
```
[   30.788592] tegra-xusb 3610000.xhci: exiting ELPG
[   30.793484] tegra-xusb 3610000.xhci: Firmware timestamp: 2019-07-24 05:47:34 UTC, Version: 60.06 release
[   30.794147] tegra-xusb 3610000.xhci: exiting ELPG done
[   30.794523] usb usb1: usb_suspend_both: status 0
[   31.045453] usb 1-2: new high-speed USB device number 2 using tegra-xusb
[   31.067911] usb 1-2: New USB device found, idVendor=2109, idProduct=2817
[   31.067922] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   31.067929] usb 1-2: Product: USB2.0 Hub             
[   31.067935] usb 1-2: Manufacturer: VIA Labs, Inc.         
[   31.067941] usb 1-2: SerialNumber: 000000000
[   31.069013] hub 1-2:1.0: USB hub found
[   31.069203] hub 1-2:1.0: 4 ports detected
[   31.191909] usb 2-3: new SuperSpeed USB device number 2 using tegra-xusb
[   31.237566] usb 2-3: New USB device found, idVendor=2109, idProduct=0817
[   31.237577] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   31.237583] usb 2-3: Product: USB3.0 Hub             
[   31.237591] usb 2-3: Manufacturer: VIA Labs, Inc.         
[   31.237596] usb 2-3: SerialNumber: 000000000
[   31.240729] hub 2-3:1.0: USB hub found
[   31.241374] hub 2-3:1.0: 4 ports detected
[   31.777466] usb 2-3: usb_suspend_both: status 0
[   31.777690] usb usb2: usb_suspend_both: status 0
[   31.797471] usb 1-2.1: new full-speed USB device number 3 using tegra-xusb
[   31.823639] usb 1-2.1: New USB device found, idVendor=17ef, idProduct=6047
[   31.823651] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   31.823658] usb 1-2.1: Product: ThinkPad Compact USB Keyboard with TrackPoint
[   31.823664] usb 1-2.1: Manufacturer: Lenovo
[   31.961458] usb 1-2.3: new high-speed USB device number 4 using tegra-xusb
[   31.983055] usb 1-2.3: New USB device found, idVendor=0bda, idProduct=8176
[   31.983066] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   31.983071] usb 1-2.3: Product: 802.11n WLAN Adapter
[   31.983077] usb 1-2.3: Manufacturer: Realtek
[   31.983081] usb 1-2.3: SerialNumber: 00e04c000001
[   32.060355] input: Lenovo ThinkPad Compact USB Keyboard with TrackPoint as /devices/3610000.xhci/usb1/1-2/1-2.1/1-2.1:1.0/0003:17EF:6047.0001/input/input5
[   32.118673] lenovo 0003:17EF:6047.0001: input,hidraw0: USB HID v1.00 Keyboard [Lenovo ThinkPad Compact USB Keyboard with TrackPoint] on usb-3610000.xhci-2.1/input0
[   32.119810] input: Lenovo ThinkPad Compact USB Keyboard with TrackPoint as /devices/3610000.xhci/usb1/1-2/1-2.1/1-2.1:1.1/0003:17EF:6047.0002/input/input6
[   32.129964] rtl8192cu: Chip version 0x10
[   32.178324] lenovo 0003:17EF:6047.0002: input,hidraw1: USB HID v1.00 Mouse [Lenovo ThinkPad Compact USB Keyboard with TrackPoint] on usb-3610000.xhci-2.1/input1
[   32.184188] rtl8192cu: Board Type 0
[   32.184336] rtl_usb: rx_max_size 15360, rx_urb_num 8, in_ep 1
[   32.184575] rtl8192cu: Loading firmware rtlwifi/rtl8192cufw_TMSC.bin
[   32.185206] ieee80211 phy0: Selected rate control algorithm 'rtl_rc'
[   32.186131] usbcore: registered new interface driver rtl8192cu
[   32.195750] usbcore: registered new interface driver rtl8xxxu
[   32.299969] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   32.306073] rtl8192cu: MAC auto ON okay!
[   32.328677] rtl8192cu: Tx queue select: 0x05
[   32.756291] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   32.856272] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   33.881747] wlan0: authenticate with 70:4f:57:be:46:f1
[   33.892941] wlan0: send auth to 70:4f:57:be:46:f1 (try 1/3)
[   33.898807] wlan0: authenticated
[   33.901480] wlan0: associate with 70:4f:57:be:46:f1 (try 1/3)
[   33.932009] wlan0: RX AssocResp from 70:4f:57:be:46:f1 (capab=0x431 status=0 aid=2)
[   33.934781] wlan0: associated
[   33.944552] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

osterwood, 2021-11-10:
Thanks for trying other hardware - that’s a great datapoint.  

USB 2 and 3 are both supported.  Did you just change the hub you were using, or also the cable?  What was the model of hub and/or cable which did not work?

vdsmax, 2021-11-11:
I used others USB 3.0 hub to test if they were working. 
The one which were working in my case are these: https://b2b.fen.pl/535bece19d6c6-usb-30/unitek-hub-usb-c-3-x-usb-31-gen-1-sd-microsd.html and https://www.bestbuy.com/site/insignia-4-port-usb-c-hub-black/6417290.p?skuId=6417290 .
The ones which was not working was this one: https://www.noelleeming.co.nz/p/endeavour-4-port-usb-2.0-hub/N141804.html
I switched many times the cables. By safety, I kept only the one for USB 3.0 connection.
Also, I noticed my keyboard was not working if it was plugged directly to the USB C connector of the board. It is working only if it's plugged on a USB 3.0 hub. I switched many time the cables to do the test. I don't think this is the issue. The keyboard I use is this one: https://support.lenovo.com/us/en/solutions/pd026745-thinkpad-compact-usb-keyboard-with-trackpoint-overview-and-service-parts

closed 2025-02-24 by pgielda