antmicro/jetson-nano-baseboard #28: Support for later versions of Jetpack?

BlueprintLab-s-hall, 2022-06-03:
Hi Ant Micro

My company has recently designed and produced a prototype custom carrier board based on your open source jetson baseboard. It has two MIPI CSI-2 ports on it that we are using to connect two Allied Vision 1800-C158C cameras. We are using a TX2 NX module in the carrier board. Unfortunately, Allied Vision's driver software is still in beta for the TX2 NX and only supports Jetpack 4.6.1 (l4t-32.7.1). I'm writing to ask whether support for Jetpack 4.6.1 (or other versions) is available somewhere or in development?
If not, is there a way we can work together to produce one?

Best regards

TJMolnar, 2022-10-28:
Hi,

Is there any update on this? I am very interested in using these boards with our Allied Vision cameras, but only if there is support for them with the current Allied Vision driver releases (Jetpack 4.6.1 or newer)

Thanks.

(tagging @tgorochowik as they committed the original patches)

tgorochowik, 2022-11-30:
We've added some patches for 32.7.2 here, for example for tx2 nx here: https://github.com/antmicro/jetson-nano-baseboard/blob/8ebbd8620282d0c63678a6a6b43d9a2abe7f230b/linux-patches/lanai-l4t-32.7.2-add-support-for-JNB-rev.1.5.1.patch

Note that this is currently the newest release that supports TX2 NX.

@TJMolnar There is no reason why this board wouldn't work with Jetpack 5.0.2 with AV cameras, but we don't have patches for that as we didn't need those yet. The boards have been used with non-av cameras and newest jetpack though.

I am closing this as there is another issue opened that asks specifically for Jetpack 5.0

closed 2022-11-30 by tgorochowik