antmicro/jetson-nano-baseboard #31: Tx2/Nano differences on carrier based on this open source board.

wild929, 2022-11-24:
Thank you for contributing to this project. It has been educational and very helpful. 

This is a long shot but I am curious if anyone has any ideas on this problem.
I designed a carrier based on this open-source board with some additional sensors. I also removed some interfaces I did not need. This board has worked reliably with the Jetson Nano module and all interfaces (HDMI, Ethernet, CSI, USB) have never presented an issue. I have 8 of them working in the field with no problems. I recently revised this PCB to include an SD card in order to work with the TX2NX.

To be brief, The Nano modules work with ver1 and ver 2 of this board. The TX2 NX modules work with ver1 but will not boot up on ver2. There are a variety of errors with some specifying emmc issues (emmc is module level and not on the carrier). 

Has anyone had any boot up issues using TX2NX modules on carriers based on this design.? Any insights on why the same TX2 modules would work with Ver 1 but not Ver 2, when they are the same except for an SD card interface?

rem0nz, 2022-11-25:
@wild929 , how can I reach you for some work ?

wild929, 2022-11-28:
You can reach me at contact@iltmicrosystems.com

Cheers!

> @wild929 , how can I reach you for some work ?

closed 2022-11-28 by wild929